pytroll/pyspectral #4: Can't find pyspectral.cfg after 'pip install pyspectral'

nebukadnezar, 2017-01-11:
On a fresh Ubuntu 16.04 box, after 'pip install pyspectral', this error occurs when using Rayleigh corrections in the true_color satpy images:

``` 
File "/usr/local/lib/python2.7/dist-packages/satpy-0.3.0-py2.7.egg/satpy/scene.py", line 656, in load
    **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/satpy-0.3.0-py2.7.egg/satpy/scene.py", line 541, in read
    return self.read_from_deptree(dataset_keys, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/satpy-0.3.0-py2.7.egg/satpy/scene.py", line 538, in read_from_deptree
    composites = self.read_composites(tree.trunk(), **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/satpy-0.3.0-py2.7.egg/satpy/scene.py", line 506, in read_composites
    **self.info)
  File "/usr/local/lib/python2.7/dist-packages/satpy-0.3.0-py2.7.egg/satpy/composites/__init__.py", line 324, in __call__
    from pyspectral.rayleigh import Rayleigh
  File "/home/balt/.local/lib/python2.7/site-packages/pyspectral/rayleigh.py", line 35, in <module>
    from pyspectral.rsr_reader import RelativeSpectralResponse
  File "/home/balt/.local/lib/python2.7/site-packages/pyspectral/rsr_reader.py", line 39, in <module>
    from pyspectral.utils import (INSTRUMENTS, download_rsr)
  File "/home/balt/.local/lib/python2.7/site-packages/pyspectral/utils.py", line 72, in <module>
    for option, value in CONF.items('general', raw=True):
  File "/usr/lib/python2.7/ConfigParser.py", line 642, in items
    raise NoSectionError(section)
ConfigParser.NoSectionError: No section: 'general'
```
Workaround is to export the location of the config file explicitly like so:

`os.environ["PSP_CONFIG_FILE"] = "/home/user/.local/etc/pyspectral.cfg"`

davidh-ssec, 2017-01-11:
We had a similar issue with satpy. I believe the solution was to move the config files listed in `setup.py` from the `data_files` list to the `package_data` list.

adybbroe, 2017-01-12:
I updated pre-master following the suggestions above, thanks @davidh-ssec .
Can you test it again using pre-master @nebukadnezar ?
Remember to unset the environment variable PSP_CONFIG_FILE

adybbroe, 2017-02-15:
Can I close this issue @nebukadnezar ?

nebukadnezar, 2017-02-15:
Yeah I think that's good now.

adybbroe, 2017-02-15:
Perfect, thanks.

closed 2017-02-15 by adybbroe