pytroll/pyspectral #11: Failing LUT dropbox links

pratheekrebala, 2017-09-10:
Looks like the dropbox links for the Rayleigh look-up tables are broken: https://dl.dropboxusercontent.com/u/37482654/pyspectral/rayleigh_only/pyspectral_rayleigh_correction_luts.tgz

This is most likely happening due to Dropbox disabling Public Links - https://www.dropbox.com/help/files-folders/public-folder

It would be awesome if you are able to share the files on another location or if I can pull them from FTP.

Thanks for the awesome work with this project!

adybbroe, 2017-09-26:
Dropbox links have been restored. Should be okay in the latest pypi release. Also, now in develop, all LUT tables and the rsr archive have been moved to Zenodo and links updated.
If you have time you are welcome testing @pratheekrebala !
So, remove your already downloaded LUTs locally and rerun. 
I expect to make a new release this week.

adybbroe, 2017-10-13:
It is solved with the latest release 0.4.3

closed 2017-10-13 by adybbroe

pratheekrebala, 2017-10-19:
Hey @adybbroe - thanks for fixing this. I just tested this - it works perfectly!

adybbroe, 2017-10-19:
Nice, I am glad to hear that!