pytroll/pyspectral #14: Examples for AHI Rayleigh corrected true color

songhan89, 2017-10-13:
Hi,

Would it be possible to have some examples how to create Rayleigh corrected true color for AHI? Thank you very much.

adybbroe, 2017-10-13:
Hi,
I have just added a jupyter notebook example here:
https://github.com/pytroll/pytroll-examples/blob/master/satpy/ahi_true_color_pyspectral.ipynb

songhan89, 2017-10-13:
Thank you very much for the quick response!

closed 2017-10-13 by songhan89

adioses, 2018-08-09:
Hi, 
Would it be possible to have the same example for GOES-16. 
I'm trying to use NetCDF files to generate Rayleigh corrected true color, I also would like to know the equations to calculate solar-zenith angle, satellite zenith angle, and the relative sun-satellite azimuth difference angle in order to understand better this corrections. Thank you very much

djhoese, 2018-08-09:
@adioses This is a rather old issue. Would you mind opening up a new issue for your specific request? Could you also include what about the AHI example isn't clear that you can't apply it to ABI (reader name is `abi_l1b`)? @adybbroe could probably link you to the paper where they discuss their Rayleigh correction. I don't have it available right now but that may answer some of your questions.

adioses, 2018-08-09:
I'm very new on GitHub and Pytroll. I'll try what you just suggest. If you could share the paper would be great. Thanks