pytroll/pyspectral #15: Adding EO-1 Hyperion Capability

patrickcgray, 2017-10-20:
Hello, I'm interested in adding the ability to process Hyperion data. Could you give point me towards any resources or explanations that would help in adding this? Thanks!

adybbroe, 2017-10-21:
Hi @patrickcgray 
I am not familiar with those data, but you are very welcome!
First you should get hold of the relative spectral response functions. I found this page:
http://www.eoc.csiro.au/hswww/oz_pi/specresp.htm
I manage to get the ALI responses but the Hyperion ZIP files are corrupted when I fetch them. Perhaps you can figure out how to get the right files?
The second step is to write a small reader/converter to put the data in hdf5. Once the data are in the internal Pyspectral hdf5 format, you are done.
You can look at some of the readers/converters already there, e.g. olci_rsr.py

adybbroe, 2018-04-23:
Hi @patrickcgray 
Have you had a chance to look further at this?
I wonder what I should do with your issue, it has now been sitting here for a while.
Best
Adam

patrickcgray, 2018-05-30:
Hey @adybbroe, sorry for the delay. I haven't had time for this and it actually has moved off of my priority list for the time being. I can close the issue if you'd like as it is no longer something I plan on doing.

adybbroe, 2018-05-30:
No worries @patrickcgray : Yes, let's close it. We can always take it up again if it should be required!