pytroll/pyspectral #28: ValueError: operands could not be broadcast together with shapes

howff, 2018-01-25:
I think this is a pyspectral error?

```
  File "/packages/pytroll/local/lib/python2.7/site-packages/satpy/scene.py", line 564, in load
    keepables = self.compute()
  File "/packages/pytroll/local/lib/python2.7/site-packages/satpy/scene.py", line 511, in compute
    return self.read_composites(nodes)
  File "/packages/pytroll/local/lib/python2.7/site-packages/satpy/scene.py", line 485, in read_composites
    self._generate_composite(item, keepables)
  File "/packages/pytroll/local/lib/python2.7/site-packages/satpy/scene.py", line 445, in _generate_composite
    keepables,
  File "/packages/pytroll/local/lib/python2.7/site-packages/satpy/scene.py", line 405, in _get_prereq_datasets
    self._generate_composite(prereq_node, keepables)
  File "/packages/pytroll/local/lib/python2.7/site-packages/satpy/scene.py", line 460, in _generate_composite
    **self.info)
  File "/packages/pytroll/local/lib/python2.7/site-packages/satpy/composites/__init__.py", line 374, in __call__
    refl_cor_band = corrector.get_reflectance(sunz, satz, ssadiff, vis.id.name, red)
  File "/packages/pytroll/local/lib/python2.7/site-packages/pyspectral/rayleigh.py", line 235, in get_reflectance
    (1 - (blueband - 20) / 80) * res)
  File "/packages/pytroll/local/lib/python2.7/site-packages/satpy/dataset.py", line 99, in wrapper
    res = func(self, other, *args, **kwargs)
  File "/packages/pytroll/local/lib/python2.7/site-packages/satpy/dataset.py", line 311, in __mul__
    return super(Dataset, self).__mul__(other)
  File "/packages/pytroll/local/lib/python2.7/site-packages/numpy/ma/core.py", line 4086, in __mul__
    return multiply(self, other)
  File "/packages/pytroll/local/lib/python2.7/site-packages/numpy/ma/core.py", line 1011, in __call__
    result = self.f(da, db, *args, **kwargs)
ValueError: operands could not be broadcast together with shapes (21440,5416) (5360,1354) 
```

Trying to load a MODIS true_color with satpy 0.8.0, pyspectral 0.5.0

mraspaud, 2018-01-25:
@howff this is a satpy issue, the wrong data is sent to pyspectral. Would you mind closing this and opening a new one in satpy ? thanks !

howff, 2018-01-25:
Created https://github.com/davidh-ssec/polar2grid/issues/144

closed 2018-01-25 by howff