pytroll/pyspectral #29: pip -e install does not copy data files

michaelaye, 2018-03-15:
I installed pyspectral locally via my GitHub fork (to fix the MPL warning)

#### Code Sample, a minimal, complete, and verifiable piece of code

```python
# Your code here
from pyspectral.solar import (SolarIrradianceSpectrum, 
                              TOTAL_IRRADIANCE_SPECTRUM_2000ASTM,
                             )
solar_irr = SolarIrradianceSpectrum(TOTAL_IRRADIANCE_SPECTRUM_2000ASTM,
                                    dlambda=0.1)
```
#### Problem description

I installed my fork into my conda environment like I always do:
```bash
pip install -e .
```
but the data file is not being copied at this point.
I know from myself that setting this up correctly in setup.py was  a pain, but I currently don't remember what was the trick.
#### Expected Output

Data should be copied correctly.

#### Actual Result, Traceback if applicable
```bash
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-93-630e5c929426> in <module>()
      1 solar_irr = SolarIrradianceSpectrum(TOTAL_IRRADIANCE_SPECTRUM_2000ASTM,
----> 2                                     dlambda=0.1)

~/miniconda3/envs/stable/lib/python3.6/site-packages/pyspectral/solar.py in __init__(self, filename, **options)

~/miniconda3/envs/stable/lib/python3.6/site-packages/pyspectral/solar.py in _load(self)

~/miniconda3/envs/stable/lib/python3.6/site-packages/numpy/lib/npyio.py in genfromtxt(fname, dtype, comments, delimiter, skip_header, skip_footer, converters, missing_values, filling_values, usecols, names, excludelist, deletechars, replace_space, autostrip, case_sensitive, defaultfmt, unpack, usemask, loose, invalid_raise, max_rows, encoding)
   1687             fname = str(fname)
   1688         if isinstance(fname, basestring):
-> 1689             fhd = iter(np.lib._datasource.open(fname, 'rt', encoding=encoding))
   1690             own_fhd = True
   1691         else:

~/miniconda3/envs/stable/lib/python3.6/site-packages/numpy/lib/_datasource.py in open(path, mode, destpath, encoding, newline)
    258 
    259     ds = DataSource(destpath)
--> 260     return ds.open(path, mode, encoding=encoding, newline=newline)
    261 
    262 

~/miniconda3/envs/stable/lib/python3.6/site-packages/numpy/lib/_datasource.py in open(self, path, mode, encoding, newline)
    614                                       encoding=encoding, newline=newline)
    615         else:
--> 616             raise IOError("%s not found." % path)
    617 
    618 

OSError: /Users/klay6683/miniconda3/envs/stable/lib/python3.6/site-packages/pyspectral/data/e490_00a.dat not found.
```
#### Versions of Python, package at hand and relevant dependencies

Using conda Python 3.6 on macOS 10.12.6, all up-to-date.

michaelaye, 2018-03-15:
Sorry, false alarm. Turns out that a change in pkg_resource paths is not dynamically reloadable, so I had to reset my IPython kernel before the paths were correctly set.

closed 2018-03-15 by michaelaye

adybbroe, 2018-03-15:
Fine! I was about to try reproduce your issue, and then it was gone in the list of open issues! :-)