pytroll/pyspectral #32: Solar interpolation, straight wavelength bins

michaelaye, 2018-03-16:
In the solar.py code you have this for interpolation in the method `interpolate`:

```python
xspl = np.linspace(start, end, (end - start) / self._dlambda)
```

Could it be that you wanted:
```python
xspl = np.linspace(start, end, round((end - start) / self._dlambda)+1)
```
because then, for a dlambda=0.001, for example, you would actually get 1 nm wide bins.
The round is required because linspace takes an integer as input there, to avoid floating point issues?

let me know, I can push another PR (but that, potentially could change some numbers on your side).

adybbroe, 2018-03-16:
Thanks, you are absolutely right, this is a bug. If you like please go ahead with the PR. 
It might affect some tests, It should at least. And if not, it indicates we need another unittest. Could be added to `test_solarflux.py`

adybbroe, 2018-03-18:
I think I fixed this issue in develop now. Can you have a look @michaelaye ?

adybbroe, 2018-04-22:
This issue should be solved by the above. No feedback from @michaelaye 
Issue being closed

closed 2018-04-22 by adybbroe