pytroll/pyspectral #45: True Color Image from Netcdf files (OR_ABI-L1b-RadF-*.nc)

adioses, 2018-08-10:
#### This is the code I'm using

```python
# 
from satpy import Scene, find_files_and_readers
from satpy.resample import get_area_def
from pyresample.utils import get_area_def

area_id = 'Sudamerica'
x_size = 3000
y_size = 1200
area_extent = (-5570248.4773392612, -4263473.5610361192, -384719.90821206354, 1339786.2707295895)
projection = '+proj=geos +a=6378137. +b=6356752.31414 +lon_0=-75 +f=.00335281068119356027 +h=35786023 +sweep=x'
description = "Sud geos"
proj_id = 'geos0'
areadef = get_area_def(area_id, description, proj_id, projection,x_size, y_size, area_extent)

scn = Scene(sensor='abi', filenames=['OR_ABI-L1b-RadF-M3C01_G16_s20182211830459_e20182211841226_c20182211841271.nc','OR_ABI-L1b-RadF-M3C02_G16_s20182211830459_e20182211841226_c20182211841260.nc','OR_ABI-L1b-RadF-M3C03_G16_s20182211830459_e20182211841226_c20182211841272.nc'])
rgbname = 'true_color'
local_scene = scn.resample(areadef)
local_scene.show(rgbname)
```
#### Problem description

I'm trying to generate a corrected true color images from GOES-16 netcdf files available en AWS.
I tried the example gave for Himawari-8 (https://github.com/pytroll/pytroll-examples/blob/master/satpy/ahi_true_color_pyspectral.ipynb). First, I couldn't use find_files_and_readers, so I define the filenames by myself. No problem on that.

Second, I couldn't process full disk image because of memory lack, I'm running the code in a 16GB RAM and 12 core server. Then I tried to process just South America. I used area definition from this web https://github.com/pytroll/satpy/blob/master/satpy/etc/areas.def

I got the following error:

Traceback (most recent call last):
  File "true_pytroll.py", line 28, in <module>
    local_scene.show(rgbname)
  File "/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/scene.py", line 986, in show
    img = get_enhanced_image(self[dataset_id].squeeze(), overlay=overlay)
  File "/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/scene.py", line 573, in __getitem__
    return self.datasets[key]
  File "/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/readers/__init__.py", line 299, in __getitem__
    key = self.get_key(item)
  File "/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/readers/__init__.py", line 288, in get_key
    best=best, **dfilter)
  File "/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/readers/__init__.py", line 239, in get_key
    raise KeyError("No dataset matching '{}' found".format(str(key)))
KeyError: "No dataset matching 'DatasetID(name='true_color', wavelength=None, resolution=None, polarization=None, calibration=None, level=None, modifiers=None)' found" 

  Thank you very much.

djhoese, 2018-08-10:
Thanks for the bug report. This seems like it is a satpy error and in the future this is probably better on the satpy repository. Anyway...

Could you try running at the top of your code `from satpy.utils import debug_on; debug_on()` and running your code again and pasting the output here. The error message you got (which I thought I had cleaned up) is saying that the true_color composite wasn't created. The debug messages should show you why.

It's possible satpy is looking for more than just C01, C02, and C03 data, but I can't think of a reason why.

adioses, 2018-08-10:
[DEBUG: 2018-08-10 11:43:08 : satpy.scene] Setting 'PPP_CONFIG_DIR' to '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc'
[DEBUG: 2018-08-10 11:43:08 : satpy.readers] Reading ['/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/avhrr_eps_l1b.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/avhrr_eps_l1b.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/avhrr_eps_l1b.yaml']
[WARNING: 2018-08-10 11:43:08 : satpy.readers.yaml_reader] No filenames found for reader: avhrr_eps_l1b
[DEBUG: 2018-08-10 11:43:08 : satpy.readers] Reading ['/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/grib.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/grib.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/grib.yaml']
[INFO: 2018-08-10 11:43:08 : satpy.readers] Cannot use ['/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/grib.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/grib.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/grib.yaml']
[DEBUG: 2018-08-10 11:43:08 : satpy.readers] while constructing a Python object
cannot find module 'satpy.readers.grib' (No module named pygrib)
  in "<string>", line 9, column 18:
        file_reader: !!python/name:satpy.readers.grib ... 
                     ^
[DEBUG: 2018-08-10 11:43:08 : satpy.readers] Reading ['/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/hdfeos_l1b.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/hdfeos_l1b.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/hdfeos_l1b.yaml']
[INFO: 2018-08-10 11:43:08 : satpy.readers] Cannot use ['/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/hdfeos_l1b.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/hdfeos_l1b.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/hdfeos_l1b.yaml']
[DEBUG: 2018-08-10 11:43:08 : satpy.readers] while constructing a Python object
cannot find module 'satpy.readers.hdfeos_l1b' (No module named pyhdf.error)
  in "<string>", line 438, column 18:
        file_reader: !!python/name:satpy.readers.hdfe ... 
                     ^
[DEBUG: 2018-08-10 11:43:08 : satpy.readers] Reading ['/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/nc_olci_l1b.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/nc_olci_l1b.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/nc_olci_l1b.yaml']
[WARNING: 2018-08-10 11:43:09 : satpy.readers.yaml_reader] No filenames found for reader: nc_olci_l1b
[DEBUG: 2018-08-10 11:43:09 : satpy.readers] Reading ['/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/mipp_xrit.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/mipp_xrit.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/mipp_xrit.yaml']
[INFO: 2018-08-10 11:43:09 : satpy.readers] Cannot use ['/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/mipp_xrit.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/mipp_xrit.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/mipp_xrit.yaml']
[DEBUG: 2018-08-10 11:43:09 : satpy.readers] while constructing a Python object
cannot find module 'satpy.readers.mipp_xrit' (cannot import name area_defs_to_extent)
  in "<string>", line 6, column 11:
      reader: !!python/name:satpy.readers.mipp ... 
              ^
[DEBUG: 2018-08-10 11:43:09 : satpy.readers] Reading ['/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/nc_olci_l2.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/nc_olci_l2.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/nc_olci_l2.yaml']
[WARNING: 2018-08-10 11:43:09 : satpy.readers.yaml_reader] No filenames found for reader: nc_olci_l2
[DEBUG: 2018-08-10 11:43:09 : satpy.readers] Reading ['/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/viirs_sdr.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/viirs_sdr.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/viirs_sdr.yaml']
[WARNING: 2018-08-10 11:43:10 : satpy.readers.yaml_reader] No filenames found for reader: viirs_sdr
[DEBUG: 2018-08-10 11:43:10 : satpy.readers] Reading ['/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/ghrsst_osisaf.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/ghrsst_osisaf.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/ghrsst_osisaf.yaml']
[WARNING: 2018-08-10 11:43:11 : satpy.readers.yaml_reader] No filenames found for reader: ghrsst_osisaf
[DEBUG: 2018-08-10 11:43:11 : satpy.readers] Reading ['/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/hrit_electrol.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/hrit_electrol.yaml', '/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/hrit_electrol.yaml']
[WARNING: 2018-08-10 11:43:11 : satpy.readers.yaml_reader] No filenames found for reader: hrit_electrol
[... cut ...]
    key = self.get_key(item)
  File "/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/readers/__init__.py", line 288, in get_key
    best=best, **dfilter)
  File "/Users/adioses/anaconda2/lib/python2.7/site-packages/satpy/readers/__init__.py", line 239, in get_key
    raise KeyError("No dataset matching '{}' found".format(str(key)))
KeyError: "No dataset matching 'DatasetID(name='true_color', wavelength=None, resolution=None, polarization=None, calibration=None, level=None, modifiers=None)' found"

djhoese, 2018-08-10:
Ah I should have realized this earlier. You need to actually load the data `scn.load(['true_color'])`.

adioses, 2018-08-10:
I got the same error Memory. It's weird because I process full disk image in the same server using GDAL. 

[DEBUG: 2018-08-10 12:06:16 : satpy.scene] Setting 'PPP_CONFIG_DIR' to '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc'
[DEBUG: 2018-08-10 12:06:16 : satpy.readers] Reading ['/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/viirs_compact.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/viirs_compact.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/viirs_compact.yaml']
[WARNING: 2018-08-10 12:06:19 : satpy.readers.yaml_reader] No filenames found for reader: viirs_compact
[DEBUG: 2018-08-10 12:06:19 : satpy.readers] Reading ['/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/nucaps.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/nucaps.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/nucaps.yaml']
[WARNING: 2018-08-10 12:06:19 : satpy.readers.yaml_reader] No filenames found for reader: nucaps
[DEBUG: 2018-08-10 12:06:19 : satpy.readers] Reading ['/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/clavrx.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/clavrx.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/clavrx.yaml']
[INFO: 2018-08-10 12:06:19 : satpy.readers] Cannot use ['/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/clavrx.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/clavrx.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/clavrx.yaml']
[DEBUG: 2018-08-10 12:06:19 : satpy.readers] while constructing a Python object
cannot find module 'satpy.readers.clavrx' (No module named pyhdf.SD)
  in "<string>", line 9, column 18:
        file_reader: !!python/name:satpy.readers.clav ... 
                     ^
[DEBUG: 2018-08-10 12:06:19 : satpy.readers] Reading ['/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/fci_fdhsi.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/fci_fdhsi.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/fci_fdhsi.yaml']
[WARNING: 2018-08-10 12:06:19 : satpy.readers.yaml_reader] No filenames found for reader: fci_fdhsi
[DEBUG: 2018-08-10 12:06:19 : satpy.readers] Reading ['/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/grib.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/grib.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/grib.yaml']
[INFO: 2018-08-10 12:06:19 : satpy.readers] Cannot use ['/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/grib.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/grib.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/grib.yaml']
[DEBUG: 2018-08-10 12:06:19 : satpy.readers] while constructing a Python object
cannot find module 'satpy.readers.grib' (No module named pygrib)
  in "<string>", line 9, column 18:
        file_reader: !!python/name:satpy.readers.grib ... 
                     ^
[DEBUG: 2018-08-10 12:06:19 : satpy.readers] Reading ['/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/avhrr_hrpt_l1b.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/avhrr_hrpt_l1b.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/avhrr_hrpt_l1b.yaml']
[INFO: 2018-08-10 12:06:19 : satpy.readers] Cannot use ['/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/avhrr_hrpt_l1b.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/avhrr_hrpt_l1b.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/avhrr_hrpt_l1b.yaml']
[DEBUG: 2018-08-10 12:06:19 : satpy.readers] while constructing a Python object
cannot find module 'satpy.readers.hrpt' (No module named pygac.gac_calibration)
  in "<string>", line 93, column 22:
            file_reader: !!python/name:satpy.readers.hrpt ... 
                         ^
[DEBUG: 2018-08-10 12:06:19 : satpy.readers] Reading ['/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/omps_edr.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/omps_edr.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/omps_edr.yaml']
[WARNING: 2018-08-10 12:06:20 : satpy.readers.yaml_reader] No filenames found for reader: omps_edr
[DEBUG: 2018-08-10 12:06:20 : satpy.readers] Reading ['/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/nc_slstr.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/nc_slstr.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/nc_slstr.yaml']
[WARNING: 2018-08-10 12:06:20 : satpy.readers.yaml_reader] No filenames found for reader: nc_slstr
[DEBUG: 2018-08-10 12:06:20 : satpy.readers] Reading ['/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/safe_sar_c.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/safe_sar_c.yaml', '/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/readers/safe_sar_c.yaml']
[WARNING: 2018-08-10 12:06:20 : satpy.readers.yaml_reader] No filenames found for reader: safe_sar_c
[... cut ...]
    self._generate_composite(item, keepables)
  File "/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/scene.py", line 712, in _generate_composite
    **self.attrs)
  File "/home/satelite/anaconda2/lib/python2.7/site-packages/satpy/composites/__init__.py", line 341, in __call__
    lons, lats),
  File "/home/satelite/anaconda2/lib/python2.7/site-packages/pyorbital-1.3.1-py2.7.egg/pyorbital/astronomy.py", line 144, in cos_zen
    return (np.sin(lat) * np.sin(dec) + np.cos(lat) * np.cos(dec) * np.cos(h__))
MemoryError

djhoese, 2018-08-10:
A couple things that may be a problem:

1. Instead of specifying `sensor` when you are creating your Scene, use `reader='abi_l1b'`. This will skip searching for a reader and will speed up the initial loading time of your script.
2. What version of satpy are you using (`satpy.__version__`)? If you aren't on 0.9 then you will probably not be able to create a full disk true color image. Version 0.9 was satpy's first version using dask and xarray which allowed us major improvements in processing time and memory usage. We provide satpy 0.9 in the `conda-forge` channel for conda if you aren't already using it, but `pip` should be fine too.
3. Do you *need* to be on python 2? This isn't really an issue for performance, but if this is a new project you are starting I highly recommend moving to python 3.

Lastly, satpy isn't simply putting the bands together. It is configured to perform multiple corrections to the data to produce the best true color image we know how to make. This includes a solar-zenith correction and a rayleigh scattering correction. The rayleigh correction requires a lot of time and memory compared to not using it. This is why gdal might not have memory issues (among other things) when creating full disk imagery. You could try loading `true_color_raw` which won't do the rayleigh correction.

I'm curious, what full disk imagery were you making with GDAL? Was it an ABI true color? How did you make the green band?

adioses, 2018-08-10:
Thank, I'll make the changes you just suggest. 
I use Satpy 0.9, I'll change to python 3 and sensor to reader.

I completely understand that, this are not simply correction and take a lot of resources. I would read more about this correction. I'm trying to create corrected true color images operational, so I need to speed up the process.

I process ABI true color. In order to produce Green Band I use a equation I found in this web (http://edc.occ-data.org/goes16/python/)
GreenB = 0.48358168 * ch2 + 0.45706946 * ch1 + 0.06038137 * ch3.
Then I make some brightness adjustment, There is not physical sense on that is just to improve visualization.

adioses, 2018-08-10:
I ran the script with CONUS images and the output is great. Thank you so much.

Also, I found an error: Platform file /home/satelite/anaconda2/lib/python2.7/site-packages/satpy/etc/platforms.txt not found. how could I correct this?

djhoese, 2018-08-10:
Don't worry about that message. It is more of a warning than an error and is actually expected in satpy right now. @adybbroe is there a solution for this platforms.txt being missing? Can't we stop the message from appearing in satpy?

djhoese, 2018-08-10:
Also FYI @adioses I run satpy on my laptop (macbook with 16GB memory) and can produce an ABI true color full disk image in ~8 minutes (maybe less, can't remember). Another thing I just noticed is that you said you have 12 cores. This may be the reason you are running out of memory since by default dask (the lower level array library we're using) will create a worker thread for each logical core on your system. You may want to configure it to have less workers, something like 8, following the directions using ThreadPool here: http://dask.pydata.org/en/latest/scheduler-overview.html#configuring-the-schedulers

adioses, 2018-08-10:
Thanks. I’ll try it. When I tried to run South America took like an hour and got out off memory.

Also I’ll try to reduce my processing area. I want to focus on South America and not in CONUS sector

djhoese, 2018-08-10:
Hhhhmmm it should never take an hour with satpy 0.9+. Make sure you have semi-recent versions of xarray, dask, pyresample, and trollimage.

adioses, 2018-08-10:
I updated all the libraries you mentioned. But I still have problems with FullDisk Images.
I couldn't configure dask. It's not really clear for me I'll keep trying.
Otherwise, I would like to know if there is a way to read just part of the file so it process less information or I should modify directly the NetCDF file?

djhoese, 2018-08-10:
Configuring the number of workers is the easiest way to limit the amount of memory used since you will only have so many workers working at a single moment. As for configuring dask, right after the imports in your script do:

```python
import dask
from multiprocessing.pool import ThreadPool
with dask.config.set(pool=ThreadPool(4)):
    # all of your other satpy-based code goes in this block of code including the .show call
```

SatPy is already attempting to minimize the amount of data being processed when you resampled the data. There is another method but it currently has a bug so I'm not going to suggest it right now.

Lastly, you can configure the "chunks" that dask creates from the data arrays. Each chunk is processed separately on an individual dask worker when possible. You can configure this right now by specifying an environment variable although we hope to make it easier in the future. You can set `PYTROLL_CHUNK_SIZE` to a value between 1024 and 4096 (default) on the command line before running your script. For bash that would be:

```bash
export PYTROLL_CHUNK_SIZE=2048
python your_script.py
```

Lastly...again, you could also provide `cache_dir='/path/where/to/put/cache/files/'` to your `resample` call. This won't fix your memory errors but it will make processing of the same geostationary area much faster on future executions.

djhoese, 2018-08-10:
What does `ls /home/satelite/Documents/GOES16/AB1_L1B/OR_ABI-L1b-RadC-M3C01_G16_s20182211832268_e20182211835041_c20182211835090.nc` give you?

What about `ls /home/satelite/Documents/GOES16/AB1_L1B/true_pytroll.py`?

adioses, 2018-08-10:
Sorry. Omit the error, I changed the files that caused the error. 
One question,  in this line 
```python 
with dask.config.set(pool=ThreadPool(4)):
```
4 is the number of cores to use?

Also, If I check the process with top command the script don't use more than one core.

djhoese, 2018-08-10:
Correct, my laptop creates 8 workers. Technically this is number of *threads* not *cores* where to me "cores" has the assumption that we're talking about multiple processes. I think you have to specify extra flags in top to get it to show you threads for a process, but that may depend on the operating system/version of top.

adybbroe, 2018-08-29:
@djhoese Can we conclude that this is rather a Satpy issue than a Pyspectral one?

adybbroe, 2018-08-29:
Okay, I close it here then. Hope the memory error is no more an issue for you @adioses

closed 2018-08-29 by adybbroe

djhoese, 2018-08-29:
@adioses Feel free to comment here or create a new issue on the satpy repository.

adioses, 2018-08-29:
Hi @djhoese, I just realice that Python version is really important. The performance of the process is incredibly better when I changed to python 3. Since I use python3 the performance allow me to process full disk in 4 minutes and do not take more than 8GB ram. Thanks for all your help