pytroll/pyspectral #58: Use dask instead of numpy masked arrays

pnuu, 2018-11-28:
Currently PySpectral uses masked arrays from Numpy to handle the data arrays. For better integration and SatPy support, these should be converted to dask arrays.

djhoese, 2018-11-28:
As mentioned on slack, there has to be a decision made about this. Does pyspectral add a hard requirement on dask? Most numpy operations should work with dask, at least the arithmetic and ufunc ones. If pyspectral can be updated and assume that NaN is used as a fill value instead of depending on masks then it should be compatible with both in most cases.

pnuu, 2018-11-28:
Running a full disk scene of SEVIRI with SatPy to create `day_microphysics` composite takes 7.4 seconds, and with the fully daskified version 4.8 s. I think we need to find a way where all the use cases can still work.

pnuu, 2018-11-29:
OK, found the way to do it: only daskify the NIR reflectance part, and adjust the documentation so that it doesn't have any references to masked arrays. If someone has used the reflectance computations without SatPy, there's a need to adjust the calling script to a) to do `.compute()` to get the result, and b) use `np.isnan(arr)` instead of `arr.mask` to get the valid values.