pytroll/pyspectral #61: can this program be used for user-defined sensor or rsr？

lwk1542, 2019-01-14:
#### Code Sample, a minimal, complete, and verifiable piece of code

```python
# Your code here

```
#### Problem description

[this should also explain **why** the current behaviour is a problem and why the 
expected output is a better solution.]

#### Expected Output

#### Actual Result, Traceback if applicable

#### Versions of Python, package at hand and relevant dependencies

Thank you for reporting an issue !

mraspaud, 2019-01-16:
In principle yes, but it would be nice of you to elaborate a bit more... What instrument, what usage of pyspectral are you planning to have ?

adybbroe, 2019-01-16:
Yes @lwk1542 could you please elaborate a bit more? The main two features provided by pyspectral and which has motivated its creation is that it allows the derivation of the reflective part of the 3.7-3.9 micron channel (of any sensor with such a band) and the atmospheric correction of the short wave visible bands for the generation of true color imagery. As also outlined in the documentation.

lwk1542, 2019-01-17:
@mraspaud @adybbroe  the COCTS（ocean color and temperature scanner）and CZI on board the new HY-1B has launched by China in 2002. The COCTS has eight visible-Near infrared bands （402nm-885nm） and two thermal bands (1030-1250nm).  The CZI has four visible-Near infrared bands. see the link for  the detail information (the ESA satellite directory: https://directory.eoportal.org/web/eoportal/satellite-missions/content/-/article/hy1b)
THANKS ！

adybbroe, 2019-01-17:
@lwk1542 Do you know where we could find the relative spectral responses for COCTS and CZI (from NSOAS)? 
Then I could add them to Pyspectral

lwk1542, 2019-01-17:
> @lwk1542 Do you know where we could find the relative spectral responses for COCTS and CZI (from NSOAS)?
> Then I could add them to Pyspectral

I can send one informal version. Can you leave an E-mail box？

adybbroe, 2019-01-17:
@lwk1542 I made a simple reader of your data to generate the pyspectral formattet hdf5 of the COCTS rsr you gave me. I can make this plot like this:
`python2 bin/composite_rsr_plot.py --platform_name HY-1C --sensor cocts --range 0.4 0.9`
![rsr_band_0040_0090](https://user-images.githubusercontent.com/169069/51330432-
cb8da980-1a77-11e9-986c-c373cba14f02.png)

adybbroe, 2019-01-17:
@lwk1542 There were no IR data. Do you have those as well?

lwk1542, 2019-01-18:
@adybbroe that is all I have.

adybbroe, 2019-01-18:
@lwk1542 Ok. I will update master branch next week with this small addition. Ok?

lwk1542, 2019-01-18:
@adybbroe of course! thank you

closed 2019-01-21 by adybbroe

adybbroe, 2019-01-21:
@lwk1542 Maybe you could just verify you can make a plot like the above. Do a git clone and use the master branch. On a linux platform:
```
git clone https://github.com/pytroll/pyspectral.git
cd pyspectral
python setup.py install --user
python bin/composite_rsr_plot.py --platform_name HY-1C --sensor cocts --range 0.4 0.9
```

PR #62 (2019-01-21): Add support for HY-1C
Adding the RSR data for HY-1C

 - [x] Closes #61 
 - [x] Tests passed 
 - [x] Passes ``git diff origin/master **/*py | flake8 --diff`` 
 - [x] Fully documented