pytroll/pyspectral #90: CREFL rayleigh Goes-16 ABI L1B

urielzan, 2019-11-06:
#### CREFL rayleigh Goes-16 ABI L1B 
Could I obtain Correction reflectance by Rayleigh in numpy array? I would like to obtain channel 2,3,5.
We found this example but it is not so clear, how can I implement every 10 minutes in the netcdf file.

https://github.com/pytroll/pytroll-examples/blob/master/pyspectral/pyspectral_rayleigh_abi.ipynb

adybbroe, 2019-11-06:
Hi @urielzan 

Yes indeed you could. In the example you found we derive the background (climatological) "rayleigh scattering" (it is not only the rayleigh scattering part but that is dominating - we also correct for atmospheric absorption by aerosols) contribution to the signal. This part should then be subtracted from the obtained uncorrected reflectance. 

Se here as well:
https://pyspectral.readthedocs.io/en/master/rayleigh_correction.html

You can do it for all bands in the 400-700 nanometer regions.

urielzan, 2019-11-06:
Thanks for your answer, We have other question:
Are there in pyspectral or another the sunz,  satz  and ssadiff  parameters calculation to obtain it for ABI L1b?
Regards

adybbroe, 2019-11-06:
@urielzan 

You would be able to get the sun-satellite viewing geometry when reading the data with Satpy.
But if I understand your quest correctly, you want to be able to read ABI level-1 data and generate  the atmosphere (Rayleigh etc) corrected reflectance. Also this can be done quite easily with Satpy (using Pyspectral underneath). Here a quick example:

```
from glob import glob
from satpy.scene import Scene
from satpy.dataset import DatasetID
from satpy.utils import debug_on
debug_on()

fl_ = glob("/home/a000680/data/GOES-16/20180114/OR_ABI-L1b-RadF*")

scn = Scene(reader='abi_l1b', filenames=fl_)

scn.load([DatasetID(name='C01', modifiers=('sunz_corrected', 'rayleigh_corrected')),
          DatasetID(name='C01', modifiers=('sunz_corrected',))])

new_scn = scn.resample(resampler='native')
dsids = new_scn.keys()
print(dsids)
print(new_scn[dsids[0]][8000:8002, 8000:8002].values)
print(new_scn[dsids[1]][8000:8002, 8000:8002].values)
```

In my case it gave this output:
```
[DatasetID(name='C01', wavelength=(0.45, 0.47, 0.49), resolution=1000, polarization=None, calibration='reflectance', level=None, modifiers=('sunz_corrected',)), DatasetID(name='C01', wavelength=(0.45, 0.47, 0.49), resolution=1000, polarization=None, calibration='reflectance', level=None, modifiers=('sunz_corrected', 'rayleigh_corrected'))]
[[17.82372238 17.82372238]
 [17.82372238 17.82372238]]
[[5.26325568 5.2656558 ]
 [5.26494942 5.26734895]]
```

urielzan, 2019-11-06:
Thank you very much, yes, it is exactly what I want to do, obtaining these bands corrected by rayleigth for the calculation of an index, I will try the code for conus.

adybbroe, 2019-11-07:
@urielzan Fine, I am glad. I think what you try doing is fine, but you should be aware that the correction we apply is purely a "climatological/view-geometry" one. It does not account for the actual aerosol load in the line of sight of the satellite of course. So, in case you want to derive information about the surface (LAI or NDVI for instance) your retrieval will of course be affected negatively in cases of events (e.g. smoke or dust) with excessive aerosol loads, just as it will also be affected by clouds if these are not removed correctly. But, hopefully using the "corrected" reflectances should give you a better input to your retrievals than the non-corrected!
All this you are probably well aware of, just wanted to make a disclaimer!
Good luck!

closed 2019-11-07 by adybbroe

simonrp84, 2019-11-07:
@urielzan Just to add to what @adybbroe says, one of the medium-term goals of my colleagues is to include the CAMS aerosol forecast, to enable a more accurate retrieval that doesn't use a climatology. It probably won't be available until the middle of 2020 at the earliest, but I thought I'd let you know.

urielzan, 2019-11-07:
Yes, I am taking into account all these considerations, for the clouds I use the product L2 ACM and the index that I try to obtain is the Index of alternative floating algae (AFAI) tested in MODIS, the index requires the correction of Rayleigth, it is only a test to know the result in ABI.

Since it is an oceanic index there may not be so much dust and smoke.

Thanks again for the prompt responses.

urielzan, 2019-11-07:
@simonrp84 ok thanks for the news, I will be aware of the publication