pytroll/pyspectral #127: AVHRR instrument naming - allow avhrr-#

adybbroe, 2021-01-25:
#### Code Sample, a minimal, complete, and verifiable piece of code

```python
# Your code here
from pyspectral.rsr_reader import RelativeSpectralResponse
avhrr3 = RelativeSpectralResponse('NOAA-19', 'avhrr-3')

```
#### Problem description

The above code sample gives a warning. The accepted naming of the sensor is `avhrr/3` and not `avhrr-3`. The former is how it is displaye on WMO Oscar, hence this the accepted naming in Pyspectral.

It would be fair though to also accept `avhrr-3`. This is also what Satpy uses, as `/` can be problematic to handle in i an yaml encoding for instance.

This means Satpy is always raising a warning here. Pyspectral anyhow fixes it so it reads the correct RSR data anyhow, so the warning in Satpy is most disturbing and confusing the user.

#### Expected Output

#### Actual Result, Traceback if applicable

#### Versions of Python, package at hand and relevant dependencies

Python 3.8
Latest Pyspectral release: 0.10.2

Thank you for reporting an issue !

djhoese, 2021-01-25:
I could see this being handled in two ways: pyspectral adds functionality for sensor "aliases" or satpy converts the sensor name before call pyspectral. I suppose doing the fix in pyspectral makes more sense long term.

adybbroe, 2021-01-25:
> I could see this being handled in two ways: pyspectral adds functionality for sensor "aliases" or satpy converts the sensor name before call pyspectral. I suppose doing the fix in pyspectral makes more sense long term.

Yes, agree. I think in this case in particular it makes sense I would say, since slashes in names are awkward to handle in code

PR #136 (2022-10-11): Fix avhrr instrument naming

 - [x] Closes #127  
 - [x] Tests added 
 - [x] Tests passed: Passes ``pytest pyspectral`` 
 - [x] Passes ``flake8 pyspectral`` 
 - [ ] Fully documented
closed 2022-10-11 by adybbroe