pytroll/pyspectral #130: Add Landsat TIRS bands

lalligagger, 2021-06-13:
The TIRS RSR is available here: https://landsat.gsfc.nasa.gov/landsat-8/spectral-response-thermal-sensor

I suggest adding to the RSR lookups, and would be happy to try to make this change however I am new to HDF5 format.

adybbroe, 2021-06-14:
That would be nice yes. We have Landsat-8 OLI supported. I am not familiar with TIRS yet. But you could try look at the convert scripts included. E.g. https://github.com/pytroll/pyspectral/blob/main/rsr_convert_scripts/oli_reader.py
I assume you will just have to copy from that one provided the original format is the same or similar. There is a general purpose hdf5 conversion function `tohdf5` included. So you basically just need to put the original data into a rather basic data object based on `InstrumentRSR`.

If you succeed you can make a PR adding the conversion script and I could ass the hdf5 data to the RSR, and upload to Zenodo. Tell me how it goes and don't hesitate if you need help.

lalligagger, 2021-06-14:
Great, thanks for the feedback! I'll try to pick this up in the next couple of weeks and let you know if I have any questions. 

It looks like learning HDF5 will be important for some other remote sensing products as well. Have been finding the package really helpful so far!

adybbroe, 2021-06-15:
Very good. Yes, HDF5 is common working with Earth Observation data. The netCDF (version 4) is, however, nowadays becoming the standard. netCDF4 is using HDF5 underneath actually. But in this case I would say you don't need to understand much of the underlying details in the HDF5 format.
I will be on vacation on and off during the summer, but please shout if you need help. Hopefully I will respond in decent time.

garnyt, 2022-06-27:
Hi there, I have created L8 TIRS and L9 OLI and TIRS rsr's in pyspectral format. Cannot attach here (h5 not allowed). Let me know where I can put these for inclusion in your library