pytroll/pyspectral #146: Could you add spectral response function for GCOM-C/SGLI sensor ?

parrot19, 2022-03-01:
Could you add spectral response function for GCOM-C/SGLI sensor ?

https://suzaku.eorc.jaxa.jp/GCOM_C/index.html

adybbroe, 2022-03-01:
Hi! 
It would be great to have GCOM-C SGLI added, indeed. I am not familiar with the instrument and do not currently work with the data, so I think it would be faster and better if you started implementing the support in Pyspectral. I will gladly try helping you!
I found RSR data here:
https://suzaku.eorc.jaxa.jp/GCOM_C/data/prelaunch/sgli_rsr_f_N_PFM201607.txt

There are also the 3-telescope data. I am not sure what that means, but I see that the instrument can scan in +-45 degrees?

I will let you judge how ambitious we should be in adding support for SGLI, and how to possibly treat the three telescopes in a general way. 

You could have a look at #144 to see how to start. Then once you have a reader for the original responses you can generated the hdf5 files and send me, or I can generate them here, and then upload to Zenodo.

parrot19, 2022-03-01:
Hi, thank you for your quick response. I think, the SGLI sensor would require two RSR to be provided from pyspectral. One for nadir view telescope and one for slant (+ 45 or -45) telescope. Unfortunately, I'm not experienced in generating the hdf5 files here....

adybbroe, 2022-03-02:
Good. Yeah, it should be fairly straight forward, and re-using the function to write hdf5, so basically it is only making a reader for the new data. Tell me if you start on something. Otherwise I will do it, but can't promise a schedule for this yet.