pytroll/pyspectral #147: On the application of spectral response function

parrot19, 2022-03-01:
I'm wondering whether someone has used this library for transferring the Sentinel-2 observation data (spectral reflectance) into the domain of MODIS by applying spectral response function of MODIS ? Is not it doable ?

adybbroe, 2022-03-01:
I am not aware of any such usage. But, we have the RSR data of both, so I suppose that is a good start. The two instruments have rather different footprint sizes, and there will be a lot of 3d effects in presence of clouds with MSI, so I suppose it is quite a challenge to what you suggest? Do you have some references to what you are after?

parrot19, 2022-03-01:
I think the whole idea is just to transfer spectral response without consideration of observation conditions. It has been discussed here, https://github.com/pmlrsg/PySpectra/commit/beb7089ac47bd0303ebcad7e754a2e693e74e0c5