pytroll/pyspectral #155: NOAA-6 spectral responses are wrong for channel 1 & 2

adybbroe, 2022-09-16:
#### Code Sample, a minimal, complete, and verifiable piece of code

When trying to plot the SW RSR data for NOAA-6 I saw that channel 1 and 2 were off, by what looks like a factor of 10:
  
[image]

```python
# Your code here
python bin/composite_rsr_plot.py --platform_name NOAA-6 --sensor avhrr -r 0.4 10.0
```

#### Problem description

I found the possible cause for this when digging out the original files from NOAA that were used to generate the internal Pyspectral RSR files. The unit change from Ångström for ch1 and 2 to `nm` for the other channels. `nm` is used for all other platforms and channels for AVHRR/1, so in the conversion script we are using a constant scaling applied to all channels and platforms to go from `nm` to microns!
 
#### Expected Output

#### Actual Result, Traceback if applicable

#### Versions of Python, package at hand and relevant dependencies

Thank you for reporting an issue !

djhoese, 2022-10-05:
This is closed with #157, right?

adybbroe, 2022-11-22:
> This is closed with #157, right?

Yes, right, fixed that now.