pytroll/pyspectral #164: Pyspectral is trying to read wrong RSR file for FY-3D/MERSI-2

pnuu, 2022-10-21:
#### Problem description
The filename of FY-3D/MERSI-2 RSR data in Zenodo is `rsr_mersi-2_FY-3D.h5`, but the latest Pyspectral 0.12.0 is trying to read `rsr_mersi2_FY-3D.h5`.

#### Expected Output
The `rsr_mersi-2_FY-3D.h5` should be read.

#### Actual Result, Traceback if applicable
[INFO: 2022-10-21 02:20:24 : satpy.modifiers.spectral] Getting reflective part of 20
[DEBUG: 2022-10-21 02:20:24 : pyspectral.rsr_reader] Filename: /home/users/satman/.local/share/pyspectral/rsr_mersi2_FY-3D.h5
[WARNING: 2022-10-21 02:20:24 : pyspectral.rsr_reader] No rsr file /home/users/satman/.local/share/pyspectral/rsr_mersi2_FY-3D.h5 on disk
[INFO: 2022-10-21 02:20:24 : pyspectral.rsr_reader] RSR data up to date, so seems there is no support for this platform and sensor

#### Versions of Python, package at hand and relevant dependencies
Python 3.10, Pyspectral 0.12.0.

pnuu, 2022-10-21:
The incoming message has the correct sensor name `mersi-2`, and I can't find where the `mersi2` is put to use in the code :thinking:

pnuu, 2022-10-21:
I added another log snippet, where the reported instrument name is `mersi-2`.

pnuu, 2022-10-21:
Ok, the instrument name is adjusted here: https://github.com/pytroll/pyspectral/blob/main/pyspectral/utils.py#L540 , introduced in https://github.com/pytroll/pyspectral/commit/290f5c20424fc3696e048d209785955c6fd73721

djhoese, 2022-10-21:
Two things:

* This line (140 of rsr_reader.py) where self.rsr.instrument = self.instrument used to be done before self._check_instrument, now it happens after: https://github.com/pytroll/pyspectral/pull/136/files?diff=split&w=0#diff-f597ae20449aa971ab1c65b2358b2f3df0d5f13da79a3584ab5a3e4b7fc3c95dR140
* The logic in line 540 of utils.py that used to be in rsr_reader.py never used to deal with hyphens. I know you mentioned this above, but I wanted to clarify that the hyphen handling is new: https://github.com/pytroll/pyspectral/pull/136/files?diff=split&w=0#diff-3cec9ec28d1b63edb89172323ae0cb4d98a45864c0dc0c3fc5df3e79a965038dR540

PR #165 (2022-10-24): Bugfix Mersi-2 RSR filename
Refactor rsr reader and tests handling the instrument name and rsr file naming in order to solve a newly introduced bug concerning the mersi-2 instrument name and its RSR filename

Now the tests pass, but the file should then be renamed on Zenodo accordingly. Right now the filename is `rsr_mersi-2_FY-3D.h5` but the code looks for a file with the name `rsr_mersi2_FY-3D.h5`. If we follow how it is done for AVHRR it should be the latter. But perhaps not overly clear that we allow the use of hyphens (`-` ) for Satellite/Platform names but not for instruments?

 - [ ] Cl [... cut]
closed 2022-10-24 by adybbroe