pytroll/pyspectral #177: Unit Convert Error in the Doc.

mqzhang1999, 2022-12-17:
#### Code Sample, a minimal, complete, and verifiable piece of code

```python
# Your code here

```
#### Problem description

On https://pyspectral.readthedocs.io/en/latest/api.html, the comment of the function pyspectral.blackbody.blackbody_wn, the unit convert "Converting from SI units to mW/m^2 sr^-1 (cm^-1)^-1: 1.0 W/m^2 sr^-1 (m^-1)^-1 = 0.1 mW/m^2 sr^-1 (cm^-1)^-1" is error . It should be 1E5, not 0.1.

#### Expected Output

#### Actual Result, Traceback if applicable

#### Versions of Python, package at hand and relevant dependencies

Thank you for reporting an issue !

adybbroe, 2023-02-16:
Thanks, you are absolutely right! Strange, since I thought we fixed this long time ago, it is correct a bit further up in the `planck` function so apparently we forgot to also update this. Thanks anyway.

adybbroe, 2023-02-16:
Ah, now I see this should have been fixed in #117 but it wasn't.

PR #180 (2023-02-16): Bugfix in function doc header
This is a small bugfix in the description of how to convert from SI units to units often used for Spectral radiance in wave number space. This should have been fixed in #117 where the same error in another doc string was fixed. But it was missed then.

 - [x] Closes #177  
 - [ ] Tests added 
 - [ ] Fully documented 
 - [ ] Add your name to `AUTHORS.md` if not there already
closed 2023-02-16 by mraspaud