pytroll/pyspectral #178: what‘s the unit of rayleigh reflectance

bodhifan, 2023-02-12:
Hi, thanks for your excellent work. I have a question: what's the unit of Rayleigh reflectance?
Here is the sample code to get Rayleigh reflectance:
-----------------------------------------------------------
>>> from pyspectral.rayleigh import Rayleigh
>>> viirs = Rayleigh('Suomi-NPP', 'viirs')
>>> import numpy as np
>>> sunz = np.array([[32., 40.], [31., 41.]])
>>> satz = np.array([[45., 20.], [46., 21.]])
>>> ssadiff = np.array([[110, 170], [120, 180]])
>>> refl_cor_m2 = viirs.get_reflectance(sunz, satz, ssadiff, 'M2')
>>> print(refl_cor_m2)
[[ 10.45746088   9.69434733]
 [ 10.35336108   9.74561515]]
------------------------------------------------------------

from the results, It is mostly like irradiance contributed by Rayleigh scattering.

adybbroe, 2023-02-15:
What you calculate is a reflection - the contribution to the observed due to Rayleigh scattering and aeorosl absorption. The units are in percent (`%`).
I realize that is not overly clear, but can be guessed from the documentation here though:
https://pyspectral.readthedocs.io/en/latest/rayleigh_correction.html

We have not come around to use pint (https://pypi.org/project/Pint/) or something like that yet, which would make this explicit.

bodhifan, 2023-02-22:
Thanks for your reply. Another question. Where is it to set aerosol parameters if Rayleigh correction relates to aerosol? Such as aerosol optical depth.