pytroll/pyspectral #179: pyspectral_rsr_data.tgz  cannot be downloaded

nothingws, 2023-02-15:
https://zenodo.org/record/7311143/files/pyspectral_rsr_data.tgz     Website cannot be downloaded

Thank you for reporting an issue !

adybbroe, 2023-02-15:
What do you mean? It is not a website but a file on the internet. When I click your above link I get the file on disk.
Try `wget -r https://zenodo.org/record/7311143/files/pyspectral_rsr_data.tgz`in a terminal for instance.
You should get this file:
`zenodo.org/record/7311143/files/pyspectral_rsr_data.tgz`

sgxl, 2023-06-20:
I have the same trouble in China, Is it possible to download the file from github?  the zenodo.org url is not reachable without VPN? or use a standby url when the zenodo.org is not reachable? thanks.

simonrp84, 2023-06-20:
We might be able to host a backup (on UK research infrastructure) but I suspect that would suffer the same blocking problems too.