pytroll/pyspectral #193: Question: Is there any details about LUT?

sgxl, 2023-09-08:
Thanks for making this awesome software open source. There are several quesions when I use it, looking for helps.

## My target
I wanna do water vapor and rayleigh correction for a visible band using pyspectral.

## My Question
I know that pyspectral can be used to do rayleigh and aerosol correction with many lookup tables.  But I am not certain about whether the water vapor absortion correction is done in the lookup tables. Is there any details about how the LUT is generated? 

1. which radiative transform model does pyspectral used, 6S? libradtran?

2. the input parameters about the LUT, does it contain the water vapor correction? If the water vapor content is set while generating the LUT, does the output of Rayleigh function contain the water vapor absortion information? 

Thanks~

mraspaud, 2023-09-08:
Thanks for you questions! I'm not expert here, but I can already point you to the scientific paper where we explain a bit about the LUT creation: https://doi.org/10.3390/rs10040560
Regarding the water vapor, I'll let others more qualified than me answer. @adybbroe @rolle-at-work

sgxl, 2023-09-08:
@mraspaud Thanks for your reply, I'll read this paper. thanks~

sgxl, 2023-09-28:
It seems that:
1. the rayleigh correction function in pyspectral has contained the correction of water vapor absorption. 
2. and it does not support a atmosphere profile: NoGaseousAbsorption, so it can not do rayleigh correction without water vapor correction.
[image]

mraspaud, 2023-09-28:
let me ping the experts: @adybbroe @rolle-at-work

mraspaud, 2023-09-29:
Here is an answer by @rolle-at-work:

> And yes, I think @sgxl is right with the conclusions.
Is there a reason to apply rayleigh correction without gas absorption?
Should we include it?

sgxl, 2023-10-07:
Thanks for your patient reply, I just saw your reply just now Because of the holiday, sorry to reply you late.

I think there is no clear need to include it. and here are my tasks for reference only,  I wanna do:
-  water vapor absortion correction with user defined water vapor value; (values from reanalysis data)
-  rayleigh correction

I think my tasks are not general tasks, so it is not neccesary to add something. I have implemented it myself, thanks.

mraspaud, 2023-10-09:
@sgxl I think personally that water vapor absorption could be an interesting addition to pyspectral if you feel like adding it :smiley:

adybbroe, 2023-10-09:
Hi @sgxl Sorry for being absent here! Yeah as the others say the method also implicitly correct for water vapour absorption - we use standard atmospheres as you have dug out yourself, and of course they comes with water vapour contribution. The overall purpose and motive of the atm correctin available so far in Pyspectral is to provide a means to correct for the strong effects of Rayleigh scattering which is not really sensitive to the actual state of the atmosphere. So, we so to speak, "remove" the background contribution, leaving the effects to the observed signal due to local and time dependent variations of the actual state of the atmosphere. The relative impact of water vapour is rather small at the shorter wavelengths (400-600nm) where the Rayleigh scattering is most pronounced. 

However, at the longer wavelengths WV absorption becomes more pronounced. What is your appplication area? Do you want to try to remove the atm absorption for the retrieval of land/sea surface parameters? 

I agree with @mraspaud that it could be a nice addition to Pyspectral if one could remove the contribution of WV absorption to the satellite observations in the VIS/NIR given an input (integrated or profile?) WV content.

closed 2024-04-15 by adybbroe