pytroll/pyspectral #220: Negative Reflectance values after atmospheric correction

akasom89, 2024-04-19:
I applied atmospheric correction to a satellite image. having negative reflectance is not reasonable based on its definition. I understand that the main reason is atmospheric data which we use for applying correction might be far away from the truth and it reflects in negative values. But I wonder what the best practice is for these negative reflectance? mapping all of them to zero!? or shifting all values towards the positive direction until all values be positive?

thanks for this awesome library.

djhoese, 2024-04-19:
I can't speak to the atmospheric correction in pyspectral and it producing negative reflectances, beyond pointing out that the rayleigh correction in pyspectral does clip it's correction from 0 to 100%.

https://github.com/pytroll/pyspectral/blob/f55b7b90992842e6969e84f4736c6c9e3f0cfe3d/pyspectral/rayleigh.py#L279

We encounter negative reflectances often in "real" satellite data for various reasons (beyond my understanding) in the Satpy package. Some readers in Satpy have added optional functionality to clip the reflectances to 0 or alternatively take the digital number (a.k.a. "count") from the instrument data that produces a positive radiance and then convert that to a reflectance. This is the most realistic for cases where an instrument recorded negative radiances/reflectances.

akasom89, 2024-04-20:
Thanks @djhoese for your answer.

akasom89, 2024-08-04:
Thanks to @djhoese for helpful answers and clarifications. 
But just for someone who has the same problem. The line mentioned in the answer is the amount of reflectance error (res), not corrected reflectance. So the amount of error due to atmosphere should be between 0 and 100. but the final corrected reflectance may still be negative.