pytroll/pyspectral #221: Brightness temperature simulation

akasom89, 2024-04-21:
I am interested to study the effect of spectral response functions of different satellites on the calculated brightness temperature. 

What I need is to provide a blackbody (in the easiest case) at an arbitrary temperature, and simulate brightness temperature of (for example) around 3.7 micron band of different sensors (Modis and ABI and …) and compare them with each other. 

As Pyspectral provides a great collection of spectral responses and also able to compute in-band solar irradiance, I thought it might be possible to handle this study using the package. I tried to make this work using documentation, but I was not successful since now. 

what I know is …
- it is possible to compute the radiance of blackbody (but not using a spectral band of sensor) 
- it is possible to compute the in-band solar_irradiance(but that is not what we can plug in to Planck rule to compute BT)

I wonder if this is possible at all or not using Pyspectral? 

Thanks in advance for this awesome library!

adybbroe, 2024-04-22:
If you assume the target is a a true blackbody at 3.7 micron the reflective part is zero, and then it is a simply convolution of the spectral response function with the Planck radiation, so for example at 280 K you can derive the radiance for the M12 and I4 bands of the NOAA-21 VIIRS like this, right?
```
from pyspectral.radiance_tb_conversion import RadTbConverter
mviirs = RadTbConverter('NOAA-21', 'viirs', 'M12')
m12rad = mviirs.tb2radiance(280.0)
iviirs = RadTbConverter('NOAA-21', 'viirs', 'I4')
i4rad = iviirs.tb2radiance(280.0)
```
I get:
```
In [11]: print(i4rad)
{'radiance': array([191065.76704347]), 'unit': 'W/m^2 sr^-1 m^-1', 'scale': 1.0}

In [12]: print(m12rad)
{'radiance': array([157540.73153277]), 'unit': 'W/m^2 sr^-1 m^-1', 'scale': 1.0}
```

adybbroe, 2024-04-22:
The you can derive the brightness temperature at any particular wavelength like this:
```
In [16]: from pyspectral.radiance_tb_conversion import radiance2tb

In [17]: radiance2tb(m12rad['radiance'], 3.7*1e-6)
Out[17]: array([279.71581719])

In [18]: radiance2tb(i4rad['radiance'], 3.7*1e-6)
Out[18]: array([283.65239919])
```

Not sure how much this helps you though?

akasom89, 2024-04-29:
Thanks @adybbroe for your thoughts. 

Also sorry for my delayed answer. 

Let me share my understanding and why I think it may not be correct  :

First step is great. the calculated radiance is based on the SRS of the sensor spectral band. I mean that is not the integration over the whole wavelengths. So we could simulate normalized radiance value of each spectral band succesfully. 

But the next step is a bit problematic. I think rad2temp does not care about the spectral band SRS at all. in other words rad2temp thinks that we are providing it the total radiance of the blackbody. 

moreover... the difference between the Brightness temperature and blackbody temperature is more than 100 degrees for spectral bands around 10 microns which seems to be wrong. don't you think so? 

Thanks a lot again for your answers.