pytroll/pyspectral #244: Seeking References for “near_infrared_reflectance”

sgxl, 2024-12-13:
Hello,

I would like to inquire whether there is any reference literature related to near_infrared_reflectance. I intend to cite it in my paper. Could you please provide some advice? Thank you.

Best regards,
Calf Lai

mraspaud, 2024-12-13:
@adybbroe ?

adybbroe, 2024-12-13:
And of course (as I guess you already know) you have the documentation on how we derive it here as well:

https://pyspectral.readthedocs.io/en/latest/37_reflectance.html

sgxl, 2024-12-13:
Thank you very much for your response!

The formulas detailed in the Pyspectral documentation are also very thorough.
I've been looking for the related papers for several days, and the references you provided are extremely helpful!

closed 2024-12-18 by adybbroe